cyberbotics/webots #3208: Node deletion and reset behavior

ad-daniel, 2021-06-21:
**Describe the Bug**
To reproduce:

- open `stewart_platform.wbt` but don't run the world.
- delete the cardboard boxes
- reset. The boxes don't return.

The behavior appears to be present both in master and 2021a, so it doesn't appear to be a regression, but I don't think it's supposed to behave like this.

lukicdarkoo, 2021-06-24:
This expected behavior (see `Adds removed nodes`):
https://cyberbotics.com/doc/reference/supervisor?version=develop#resetreload-matrix

I am even using it as a feature :)

ad-daniel, 2021-06-24:
You learn something new everyday, that table is quite useful actually. Should use the develop version of the docs more often, thanks!

closed 2021-06-24 by ad-daniel