cyberbotics/webots #3214: Webots crash in combination with connected Autoref. (RoboCup Humanoid-League)

MaximilianGiessler, 2021-06-23:
**Describe the Bug**
Bug-report: Team Sweaty with Webots and autoref.

Webots crashes every time after beaming our robot, if for example we standing on a illegal position in mode "SET".

Webots crashes with a "memory access error" and this also happens when our docker is not running 

**Steps to Reproduce**
When reaching the SET state, it checks if all robot have a valid position (inside their own half). When our software leads to wrong positioning, the robot is beamed to the reentyStartingPose and Webots/Autoref is crashing. Again without any errors in logs. 

**Expected behavior**
The robot should be beamed to the "reentyStartingPose" and webots should run stable.

**Screenshots**
Log-File:

"[0000.002|0000.000] Warning: No players found for team HERoEHS
[0000.002|0000.000] Info: Minimum real time factor is set to 0.
[0000.002|0000.000] Info: Simulation will run as fast as possible, real time waiting times will be minimal.
[0002.892|0000.000] Info: Spawned RED_PLAYER_1 Sweaty3 on port 10001 at halfTimeStartingPose: translation (-5.0 -4.8 0.85), rotation (0 0 1 0).
[0002.901|0000.000] Info: Connected to GameControllerSimulator at localhost:8750.
[0002.902|0000.000] Info: Game type is NORMAL.
[0002.902|0000.000] Info: Red team is "Sweaty", playing on left side.
[0002.902|0000.000] Info: Blue team is "HERoEHS", playing on right side.
[0002.902|0000.000] Info: Sending 1:SIDE_LEFT:27 to GameController.
[0002.902|0000.000] Info: Waiting for GameController to answer to 1:SIDE_LEFT:27.
[0003.102|0000.000] Info: Waiting for GameController to answer to 1:SIDE_LEFT:27.
[0003.303|0000.000] Info: New state received from GameController: STATE_INITIAL.
[0003.303|0000.000] Info: State has succesfully changed to INITIAL
[0003.303|0000.000] Info: New secondary state received from GameController: STATE_NORMAL, phase 0.
[0003.305|0000.000] Info: Regular start: Waiting 5 seconds (real-time) before going to READY
[0003.305|0000.000] Info: Kick-off is red.
[0003.305|0000.000] Info: Moved ball out of the field temporarily
[0003.305|0000.000] Info: Ball not in play, will be kicked by a player from the red team.
[0003.305|0000.000] Info: Sending 4:KICKOFF:27 to GameController.
[0003.305|0000.000] Info: Waiting for GameController to answer to 4:KICKOFF:27.
[0003.505|0000.000] Info: Waiting for GameController to answer to 4:KICKOFF:27.
[0008.311|0004.224] Info: Real-time to wait elasped, moving to READY
[0008.311|0004.224] Info: Sending 536:STATE:READY to GameController.
[0008.311|0004.224] Info: Waiting for GameController to answer to 536:STATE:READY.
[0008.511|0004.224] Info: Waiting for GameController to answer to 536:STATE:READY.
[0008.712|0004.224] Info: New state received from GameController: STATE_READY.
[0008.712|0004.224] Info: State has succesfully changed to READY
[0023.724|0018.312] Info: STATUS: Avg speed factor: 0.915 (over last 20.00 seconds)
[0023.724|0018.312] Info: STATUS: state: STATE_READY, remaining time: 600
[0043.729|0037.072] Info: STATUS: Avg speed factor: 0.938 (over last 20.01 seconds)
[0043.729|0037.072] Info: STATUS: state: STATE_READY, remaining time: 600
[0056.711|0049.272] Info: New state received from GameController: STATE_SET.
[0056.717|0049.272] Info: Waiting for classic play
[0056.718|0049.272] Info: Ball respawned at 0 0 0.1125.
[0056.734|0049.280] Info: Sending 6171:PENALTY:27:1:INCAPABLE to GameController.
[0056.734|0049.280] Info: Waiting for GameController to answer to 6171:PENALTY:27:1:INCAPABLE.
[0056.935|0049.280] Info: Waiting for GameController to answer to 6171:PENALTY:27:1:INCAPABLE.
[0057.135|0049.280] Info: placing player red 1 at [-4.9, -4.8, 0.85]
[0057.135|0049.280] Info: -> pos: [-4.9, -4.8, 0.85]
[0057.140|0049.280] Info: Red player 1 reset to None: translation (-4.9 -4.8 0.85), rotation (0 0 1 0).
[0057.140|0049.280] Info: Disabling actuators of red player 1.
[0057.140|0049.280] Info: INCAPABLE penalty for red player 1: outside of field at kick-off."

**System**
 - Operating System: [Linux Ubuntu 20.04]

**Additional context**
Add any other context about the problem here.

MaximilianGiessler, 2021-06-23:
Webots crashes in the next step after the reset_player function was called.

omichel, 2021-06-23:
It could be that the proto of the robot doesn't support well the regeneration. I would try to turn this proto into a non-regenerated proto to see if that fixes the problem. See https://github.com/RoboCup-Humanoid-TC/webots/issues/154#issuecomment-855389662 for an explanation.

MaximilianGiessler, 2021-06-23:
Unfortunately, the tip did not lead to the desired result ....

omichel, 2021-06-23:
I found the problem...
In `Sweaty3Robot.proto`, you should remove line 14: `local dummy = fields.customData` which is very odd and add around line 32 `customData IS customData`.  I don't know yet exactly why it crashes, but this change will fix the problem and it is the way to go.

MaximilianGiessler, 2021-06-23:
Thank you, that fixed the crash!

closed 2021-06-23 by MaximilianGiessler