cyberbotics/webots #3220: Lidar crash

omichel, 2021-06-23:
Some Lidar configuration cause Webots to crash (tested on the develop branch).
Open the [projects/samples/devices/worlds/lidar.wbt](https://github.com/cyberbotics/webots/blob/develop/projects/samples/devices/worlds/lidar.wbt) in Webots in pause mode and remove the `Lidar.rotatingHead` field and set the `Lidar.fieldOfView` to `1`. Save and revert &rarr; crash. If you attempt to open this world, it will crash when running it as well.

lukicdarkoo, 2021-06-24:
I cannot reproduce this issue, I tried even to be more extreme with the `Lidar.fieldOfView`, but it doesn't crash

omichel, 2021-06-25:
It definitely crashes on my Windows machine...

llessieux, 2021-10-01:
I cannot reproduce the crash but when I set the Lidar.fieldOfView to 1, I am losing all the rays. Not sure if that is expected or not ;)
I am using Windows too (11).

omichel, 2021-10-01:
Now, I cannot reproduce the crash from the master branch either (on Windows 10).

> but when I set the Lidar.fieldOfView to 1, I am losing all the rays. Not sure if that is expected or not ;)

This is not expected and I cannot produce this problem: I start the simulation and the value of the Lidar.fieldOfView is 1.57, I can run the simulation, set it to 1, and I still see the rays. Or I can stop the simulation, reload it, set the Lidar.fieldOfView to 1, save, revert and run and I can still see the rays. Under which circumstances do you see the rays disappearing?

llessieux, 2021-10-02:
Strange, I updated the source, rebuilt it and I can't reproduce that anymore.
Not sure what happened.

omichel, 2021-10-04:
I am closing this issue as the crash is not reproducible any more.

closed 2021-10-04 by omichel