cyberbotics/webots #3232: It‘s time to add pointcloud2 topic for velodyne lidar

JackyMao1999, 2021-06-25:
system version: Ubuntu 20.04
Webots version: Webots 2021a

Now, ROS has abandoned sensor_msgs/PointCloud topic, and both loam and cartographer algorithms use sensor_msgs/PointCloud2 topic.
If you can accept this question,thank you very much, if can't accept this issue, how to publish the multi line lidar data with pointcloud2 type.  

Thanks for your reading.

lukicdarkoo, 2021-06-25:
We are using the PointCloud2 messages in our ROS 2 interface:
https://github.com/cyberbotics/webots_ros2/blob/feedcb5293fc717b41db177eadc56a8b82b50d3c/webots_ros2_driver/src/plugins/static/Ros2Lidar.cpp#L57-L75

We will be happy to review a backport to the ROS 1 interface :)

JackyMao1999, 2021-06-26:
wow, thank you.

closed 2021-11-15 by BenjaminDeleze