cyberbotics/webots #3248: Deprecated warning in highway_driving benchmark

stefaniapedrazzi, 2021-06-30:
This warning is now displayed in the `highway_driving` benchmark:
```
Warning: wb_supervisor_node_get_number_of_contact_points() is deprecated, use wb_supervisor_node_get_contact_points() instead.
```

The controller should be updated to use the new methods.

stefaniapedrazzi, 2021-06-30:
Other simulations also needs to be updated, for example the `obstacle_avoidance` benchmark.

ad-daniel, 2021-06-30:
does it also need to be removed from the test suite or just when it is no longer supported? Currently the test checks both the old and new method

stefaniapedrazzi, 2021-06-30:
> does it also need to be removed from the test suite or just when it is no longer supported? Currently the test checks both the old and new method

It is ok to keep both in the tests.

PR #3256 (2021-07-01): Remove deprecated functions from samples
**Description**
Fixes #3248. I didn't find other instances.
closed 2021-07-01 by mael25