cyberbotics/webots #3258: P-Rob3 device names

stefaniapedrazzi, 2021-06-30:
In `P-Rob3` PROTO models, the device names are not very self-explanatory but they use names likes  `1`, `2`, etc. (for joints), `2 sensor`, `2 sensor` (position sensors), etc.
Then for buttons the style changes and the number comes after: `button 1`, `button 2` (same for distance sensors)

Is there a reason for this choice?
Especially using numbers only like for the joints doesn't seem user-friendly and when seeing it in the web robot component I think the first impression is that there are some bugs.
The it would be nice to have some coherence in the device name format at least inside the same model.

mael25, 2021-07-01:
I do not know the reasons for this choice (it is not part of the new robots introduced)... But I agree that it could have more self-explanatory names, I will look into that!

stefaniapedrazzi, 2021-07-01:
Sorry, I though it was one of the new ones.
I will check with the specifications if there is some reasons for this names.

But this can be postpones given that it is not something new.

stefaniapedrazzi, 2021-07-12:
Note that there are still format mismatch:

On onide the number is at the end: `button 1`, `motor 2`.
But for sensor we use: `motor 1 sensor`.

To me this is not consistent and would have been better to choose `motor sensor 1`.

omichel, 2021-07-12:
> 
> 
> Note that there are still format mismatch:
> 
> On onide the number is at the end: `button 1`, `motor 2`.
> But for sensor we use: `motor 1 sensor`.
> 
> To me this is not consistent and would have been better to choose `motor sensor 1`.

We usually name these sensors with the name of the motor, plus a space or an underscore and the word "sensor", which explains this choice, see for example "palm_finger_1_joint_sensor" in `Robotiq3fGripper.proto` or "wrist_3_joint_sensor" in `UR10e.proto`. Also, in `Servo.proto`, the name of the sensor is computed this way: `name %<= '"' + fields.name.value + ' sensor"' >%`.

stefaniapedrazzi, 2021-07-12:
Yes.
But in this case the devices names are not consistent within the robot definition.
Even the motor name should rather be `hinge 1 motor` instead of `motor 1` if we would have some consistency.

omichel, 2021-07-12:
> Even the motor name should rather be hinge 1 motor instead of motor 1 if we would have some consistency.

I believe it depends on the real robot. The name we have in Webots should be as close as possible to the names of the real robot. This is what we did for example with the Aibo and Nao robots, where names are like "PRM:/r3/c1-Joint2:31" or "RFoot/Bumper/Right", e.g., not consistent with other robots. For the P-rob3, motors are named this way (motor 1, motor 2, etc.). Note also some inconsistency because motor names are starting from 1 while button names are stating from 0. But changing this would confuse the users of the real robot.

PR #3390 (2021-07-09): Fixed P-Rob3 motor names
Fixes #3258.
closed 2021-07-12 by stefaniapedrazzi