cyberbotics/webots #3268: icon_creator: controller is crashing

stefaniapedrazzi, 2021-07-01:
On my Ubuntu 18.04 system, the `icon_creator` controller is crashing when processing the image to auto-crop it and remove the background.

mael25, 2021-07-21:
I tried on my virtual machine (Ubuntu 18.04.5 LTS) with R2021b-rev1 but I cannot reproduce the issue...

stefaniapedrazzi, 2021-07-21:
I checked on a Ubuntu 20.04 VM and it is also working properly.

stefaniapedrazzi, 2021-07-21:
It seems that the problem is the version of the `pillow` library:
* pillow 7.0.0 works correctly
* pillow 8.1.0 doesn't work (previously installed on my development system)

stefaniapedrazzi, 2021-07-21:
In Pillow 7.1.0 the behavior of `Image.getbbox` changed:
https://pillow.readthedocs.io/en/stable/releasenotes/7.1.0.html#if-present-only-use-alpha-channel-for-bounding-box

PR #3482 (2021-07-21): Fix icon_studio controller with Pillow >= 7.1.0
Fix #3268:
`getbbox()` behavior with RGBA images changed in Pillow 7.1.0 and broke the `icon_studio` controller.
closed 2021-07-21 by stefaniapedrazzi