cyberbotics/webots #3274: Controller crashes when deleting robot in display.wbt

stefaniapedrazzi, 2021-07-01:
The following error messages are printed in the console after deleting the robot in `display.wbt`:
```
WARNING: display_supervisor: The process crashed some time after starting successfully.
display_supervisor: request.c:181: request_is_over: Assertion `r->pointer <= request_get_size(r)' failed.
WARNING: 'display_supervisor' controller crashed.
```

It would be expected that the controller is terminated without errors.

lukicdarkoo, 2021-07-01:
I am not getting that error but:
```
WARNING: SolidBox (PROTO) > DEF SQUARE_ARENA_APPEARANCE_A PBRAppearance > ImageTexture: 'webots://projects/default/worlds/textures/metal.jpg' not found.
A resource file can be defined relatively to the worlds directory of the current project, relatively to the worlds directory of the default project, relatively to its protos directory (if defined in a PROTO), or absolutely.
```

stefaniapedrazzi, 2021-07-01:
Note that if you want to test in the same environment as in the release package you should first run `scripts/package/update_worlds_and_protos.py`.

Otherwise make sure that the world you are opening is located in the same folder as the Webots main application.

ad-daniel, 2021-07-06:
I can't seem to reproduce it on ubuntu 20.

BenjaminDeleze, 2021-07-06:
I can reproduce it on Ubuntu 20 (delete the robot named MYBOT, not the one named robot)

PR #3375 (2021-07-08): fix crash controllers
**Description**
Fix #3274 

**Tasks**
Add the list of tasks of this PR.
  - [x] Fix the crash on delete, introduced in #2279 
  - [x] Fix the `Warning: QObject::disconnect: Unexpected nullptr parameter`, introduced in #3237
closed 2021-07-08 by BenjaminDeleze