cyberbotics/webots #3277: Nao room in guided tour is failing

omichel, 2021-07-02:
The `nao_room.wbt` launched from the guided tour is failing with the following error (on Windows):
```
WARNING: Background: Texture dimension mismatch between leftUrl and rightUrl.
WARNING: "python.exe" was not found.
Webots requires Python version 3.9, 3.8, 3.7 or 2.7 (64 bit) from python.org in your current PATH.
To fix the problem, you should:
1. Check the Python command set in the Webots preferences.
2. Check the COMMAND set in the [python] section of the runtime.ini file of your controller program if any.
3. Fix your PATH environment variable to use the required Python 64 bit version (if available).
4. Install the required Python 64 bit version and ensure your PATH environment variable points to it.
```

lukicdarkoo, 2021-07-02:
On macOS as well. However, adding the full path to the Python executable solves the problem. It seems that Webots ignores the PATH environment somehow

omichel, 2021-07-02:
Anyhow, the guided tour should not have any dependency on Python, as it is not installed by default on Windows.

PR #3323 (2021-07-05): Fix nao room demo
Fixes  #3277.
closed 2021-07-05 by omichel