cyberbotics/webots #3281: sumo_interface_example.wbt crashes Webots

omichel, 2021-07-02:
Running `projects/vehicles/worlds/sumo_interface_example.wbt` display the following warnings:
```
INFO: sumo_supervisor: Starting controller: python.exe -u sumo_supervisor.py --max-vehicles=100 --rotate-wheels --port=8873 --seed=1 --step=200 --radius=-1 --maximum-lateral-speed=2.5 --maximum-angular-speed=3 --lane-change-delay=3
INFO: enable_all_lidars: Starting controller: python.exe -u enable_all_lidars.py
Temporary network files generated in C:\Users\Olivi\AppData\Local\Temp\tmpyafao2qc
Loading configuration... done.
Parsing nodes from 'C:\Users\Olivi\AppData\Local\Temp\tmpyafao2qc\sumo.nod.xml'... done.
Parsing edges from 'C:\Users\Olivi\AppData\Local\Temp\tmpyafao2qc\sumo.edg.xml'... done.
Parsing connections from 'C:\Users\Olivi\AppData\Local\Temp\tmpyafao2qc\sumo.con.xml'... done.
 Import done:
   9 nodes loaded.
   16 edges loaded.
Removing self-loops... done (0ms).
Removing empty nodes... done (0ms).
   0 nodes removed.
Moving network to origin... done (0ms).
Computing turning directions... done (0ms).
Sorting nodes' edges... done (0ms).
Computing node shapes... done (1ms).
Computing edge shapes... done (0ms).
Computing node types... done (0ms).
Computing priorities... done (0ms).
Computing approached edges... done (0ms).
Computing approaching lanes... done (0ms).
Dividing of lanes on approached lanes... done (0ms).
Processing turnarounds... done (0ms).
Rechecking of lane endings... done (0ms).
Assigning nodes to traffic lights... done (0ms).
Computing traffic light control information... done (0ms).
Computing node logics... done (0ms).
Computing traffic light logics... done (0ms).
 1 traffic light(s) computed.
Building inner edges... done (0ms).
-----------------------------------------------------
Summary:
 Node type statistics:
  Unregulated junctions       : 0
  Priority junctions          : 9
  Right-before-left junctions : 0
 Network boundaries:
  Original boundary  : -510.00,-510.00,510.00,510.00
  Applied offset     : 510.00,510.00
  Converted boundary : 0.00,0.00,1020.00,1020.00
-----------------------------------------------------
Writing network... done (3ms).
Success.
Connect to SUMO... This operation may take a few seconds.
WARNING: enable_all_lidars: The process crashed some time after starting successfully.
WARNING: 'enable_all_lidars' controller crashed.
```

And finally, Webots crashes.

mael25, 2021-07-02:
It also crashes with a segmentation fault on Ubuntu 20.04 (Snap)

ad-daniel, 2021-07-02:
The culprit appears to be `lidar.getPointCloud()`, but only for `data_type='list'` (which is the default), for `data_type='buffer'` it doesn't crash.

closed 2021-07-06 by ad-daniel