cyberbotics/webots #3283: Momentary white textures when regenerating or resetting

ad-daniel, 2021-07-02:
**Describe the Bug**
To reproduce:
- open `mantis.wbt`
- reset.

Or:
- add arena
- change arena size.

In either case, after doing so the textures are white for like 1s before returning.
I imagine it's related to the fact that it has to check whether the server has a newer version of the textures. I don't know if it's something that can be avoided.

**System**
 - Operating System: Windows

lukicdarkoo, 2021-07-02:
Also very visible in `bb-8.wbt`

BenjaminDeleze, 2021-07-05:
Yes it is normal, because we launch the downloads of textures but we do not wait for them to finish to render the corresponding object. That is why it appears white for a short moment. 

If we are downloading for more than 1 second, a popup will appear to show the progress bar.

I think it will be complicated to avoid this short moment of textureless objects.

ad-daniel, 2021-07-05:
While loading a world for the first time, a blank screen that says `loading world` shows up. Would it make sense to display something similar on a reset until all the assets are available?

BenjaminDeleze, 2021-07-05:
I think that the actual solution (display a pop up after 1sec) is fine and that displaying a back screen would reduce the fluidity of the reset, but this is debatable, maybe it is more important to hide the textureless objects.

> EDIT: actually it might no longer be an issue if we can avoid checking if the cache is up to date at every reset.

Unfortunately not. It just tested it in offline mode (so always-cache) and there is a short moment where it is still possible to see objects without textures.

ad-daniel, 2021-07-05:
For a normal user the popup might be sufficient, but if somebody is doing some types of learning where they continuously reset (possibly on a headless system and having the world running on fast mode), would a camera/sensor pick up this missing texture during that second they're missing? If so, it's important to prevent the simulation from running until the scene is fully ready.

PR #3349 (2021-07-06): Improve Urls handling
**Description**
Several improvements to webots when worlds use URLs

**Tasks**
  - [x] Pause the simulation until all the textures are ready (on reset for example), fix #3283 
  - [x] Change the default cache policy from `prefer network` to `prefer cache`.
  - [x] Modify the expiration date to be one day. (default = 5min).
closed 2021-07-06 by BenjaminDeleze