cyberbotics/webots #3295: Robot window is not populated after reloading a simulation

omichel, 2021-07-02:
**Steps to Reproduce**
1. Open `projects/languages/worlds/python.wbt`
2. Stop the simulation if it is running.
3. Double-click on a robot to open its robot window.
4. Save the simulation and revert.
5. The robot window is empty, it will populate only if we press the step or run button.

**Expected behavior**
The robot window should not be empty after reload.
It is not empty when opening the world in "pause" mode, which is correct.

**System**
 - Operating System: Windows 10

ad-daniel, 2021-07-08:
On ubuntu 20, opening this robot window in pause mode is empty. It gets populated only after the first time step.
This isn't true for other worlds though. The robotbenchmark ones for instance it starts already populated.

stefaniapedrazzi, 2021-07-12:
This behavior is not new and has always been like this for the generic robot window: after opening it you need at least one simulation step to get the devices content.
The main reason is that in the generic robot window it is the robot window itself that notifies the controller that the devices information are needed and these are only generated and sent in the next simulation step.

This is different in the robotbenchmark simulations, where messages are sent from the controller to the robot window independently if the robot window is open or not (not that this won't be efficient in generic case).

So, it is not possible to have immediately a populated robot window when opening it in pause mode.
If needed, we can add a message explaining that a simulation step is needed to populate it.

omichel, 2021-07-12:
What I don't quite understand is that after following the above procedure to reproduce the issue on my Windows machine, if start Webots in pause mode (`webots --mode=pause`), the robot window will get populated immediately on start-up (no need to press step or run). Then, if I press revert, the robot window will be empty. That seems odd and look like a bug.

stefaniapedrazzi, 2021-07-12:
I cannot reproduce this behavior on my machine Ubuntu 18.04 (as already reported by @ad-daniel  for Ubuntu 20.04).
When I start Webots in pause mode, the robot window is not populated as well.
So I'm getting the exact same behavior as on reset.

We should maybe check if on Windows an additional step is automatically executed on start/load.

omichel, 2022-06-24:
I cannot reproduce this bug anymore with the robot window in the browser.

closed 2022-06-24 by omichel