cyberbotics/webots #3307: ROS libraries

mael25, 2021-07-02:
**Describe the Bug**
When launching the `e_puck_line.launch` ros example, I get the following error:

```
/snap/webots/x1/usr/share/webots/projects/default/controllers/ros/ros: error while loading shared libraries: libboost_system.so.1.65.1: cannot open shared object file: No such file or directory
WARNING: 'ros' controller exited with status: 127.
```

I had to manually add the ros libraries to the `LD_LIBRARY_PATH`:

export LD_LIBRARY_PATH=${WEBOTS_HOME}/lib/controller:`${WEBOTS_HOME}/projects/default/controllers/ros/lib/ros`

As I have not seen in the tutorial that it was needed to do so, I wonder if this is normal.

**System**
 - Ubuntu 20.04, Snap version of Webots

lukicdarkoo, 2021-07-02:
The missing `libboost_system.so.1.65.1` library was reported by a Discord user, so it seems to be a real problem. Did you use the ROS package from apt or you compiled it from develop branch?

mael25, 2021-07-02:
From apt

lukicdarkoo, 2021-07-02:
This works for me:
```
$ source /opt/ros/noetic/local_setup.bash
$ export WEBOTS_HOME=/home/lukic/Downloads/webots-R2021b-x86-64/webots
$ roslaunch webots_ros e_puck_line.launch

# the simulation works

$ echo $LD_LIBRARY_PATH
/opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu
$ echo $PYTHONPATH
/opt/ros/noetic/lib/python3/dist-packages
$ echo $WEBOTS_HOME
/home/lukic/Downloads/webots-R2021b-x86-64/webots
```

lukicdarkoo, 2021-07-02:
I managed to reproduce it with Webots snap

PR #3439 (2021-07-15): Add ROS and snap usage documentation
Fixes #3307, replaces (https://github.com/cyberbotics/webots-snap/pull/34)

https://cyberbotics.com/doc/guide/tutorial-8-using-ros?version=documentation-ros-snap
closed 2021-07-02 by lukicdarkoo