cyberbotics/webots #3310: Python example controllers crash

mael25, 2021-07-02:
**Describe the Bug**
The slave and driver controller crashes in `projects/languages/python/worlds/examples.wbt` with:

Traceback (most recent call last):
  File "slave.py", line 21, in <module>
    from controller import AnsiCodes
  File "/usr/local/webots/lib/controller/python27/controller.py", line 2330
    print("Error: `data_type` cannot be `{}`! Supported values are 'list' and 'buffer'.".format(data_type), file=sys.stderr)
                                                                                                                ^
SyntaxError: invalid syntax
WARNING: 'slave' controller exited with status: 1.

**System**
 - Ubuntu 20.04 (Webots Deb)

lukicdarkoo, 2021-07-02:
This Python 2.7, I am not sure whether we want to support it anymore

omichel, 2021-07-02:
No, we don't want to support Python 2.7 any more.

omichel, 2021-07-02:
We should probably get rid of the python2.7 library.

mael25, 2021-07-02:
It is weird because Python 3 was also installed on my machine...

mael25, 2021-07-02:
When I tried with the Snap installation earlier it was using python3 automatically, but not with the Debian one

PR #3313 (2021-07-02): Drop Python 2.7
Python 2.7 has been deprecated for long now. We should get rid of it.
Fixes #3310.
closed 2021-07-02 by omichel