cyberbotics/webots #3324: Name changed by path in proto

mael25, 2021-07-05:
**Describe the Bug**
The `name` field of two components has been changed as if it were a path to a resource in `Fabtino.proto`. This might have occured in other protos.
```

device [
          RotationalMotor {
            name "webots://projects/robots/rec/fabtino/protos/front_right_wheel_joint"
            maxVelocity 100
            maxTorque 100
          }
          PositionSensor {
            name "webots://projects/robots/rec/fabtino/protos/front_right_wheel_joint_sensor"
          }
        ]
```

mael25, 2021-07-05:
Thus the controller does not work in `fabtino.wbt` #3321

ad-daniel, 2021-07-05:
Yes, kinda surprising it didn't affect all other worlds as well. These fields were manually converted?

BenjaminDeleze, 2021-07-05:
Yes, as the Fabtino was added after all the worlds/protos were already converted, I just changed its fields manually. It must be a ctrl-c, ctrl-v error.

closed 2021-07-05 by mael25