cyberbotics/webots #3332: Javascript error in `custom_robot_window_simple.wbt`

ad-daniel, 2021-07-05:
**Describe the Bug**
The message only appears when resetting.
`[javascript] ReferenceError: Can't find variable: QWebChannel (webots.js:65)`

omichel, 2021-07-06:
I cannot reproduce this error on my Windows machine with the latest nightly build.

BenjaminDeleze, 2021-07-06:
I cannot reproduce it with the latest nightly build on Ubuntu 20

ad-daniel, 2021-07-06:
I still have it on the latest Windows nightly.

ad-daniel, 2021-07-06:
By manually copying the file `qwebchannel.js` from a linux build into `resources/web/local` and adding `<script src='resources/web/local/qwebchannel.js'></script>` to `custom_robot_window_simple.html` (and therefore bypassing [this instruction](https://github.com/cyberbotics/webots/blob/8e35fb8d7bc8217470a5e680b5b0d3108a56b739/src/webots/gui/WbRobotWindow.cpp#L121)) the warning disappears.

What I don't understand however is where it is supposed to find this resource in the first place, for linux and mac this is included in the package [here](https://github.com/cyberbotics/webots/blob/8e35fb8d7bc8217470a5e680b5b0d3108a56b739/scripts/packaging/files_core.txt#L333) here and copied from [here](https://github.com/cyberbotics/webots/blob/8e35fb8d7bc8217470a5e680b5b0d3108a56b739/scripts/install/qt_linux_installer.sh#L115-L116). Where it managed to find it for other Windows machines for it to affect only my machine I have no idea.

PR #3370 (2021-07-07): Fixed robot window javascript error on reset
Fixes #3332.
closed 2021-07-07 by omichel