cyberbotics/webots #3353: sample world mir100.wbt sick_s300glass.dae cannot download

Thomas-Feldmeier, 2021-07-06:
ubuntu 20.04, webots nightly build jul 5, sample world mir100:
`WARNING: Mir100 (PROTO) > SickS300 (PROTO) > Shape > Mesh: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/devices/sick/protos/meshes/sick_s300_glass.dae: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/devices/sick/protos/meshes/sick_s300_glass.dae`

ad-daniel, 2021-07-06:
I just tried on the latest nightly (windows) and it worked. You're using snap, tarball or deb on ubuntu?

BenjaminDeleze, 2021-07-06:
It seems like the urls are badly converted for the nightly build on ubuntu. It should be `https://raw.githubusercontent.com/cyberbotics/webots/<commit id>/...` and not `https://raw.githubusercontent.com/cyberbotics/webots/R2021b/... `

However the nightly seems well built on windows

PR #3359 (2021-07-06): Fix nightly urls
**Description**
Fix #3353 

- [x] Fetch the tags beforehand to be sure to have them locally
- [x] Check if we are currently on a tag or not
closed 2021-07-06 by omichel