cyberbotics/webots #3377: Blender import not working

Majanao, 2021-07-08:
Hi, 

I'm trying to import robots and also more simpler models like a simple cube from a Blender file to Webots. I always get the error:
_WARNING: Invalid data, please verify mesh file (bone weights, normals, ...): BLEND: Expected at least one object with no parent_

I tried a bunch of different models and different setups, but none of them worked, Unfortunately the blender-webots-exporter doesn't work with Blender 2.9 anymore.

Can anyone report a successful use of this function?

omichel, 2021-07-08:
Are you using the [Mesh](https://cyberbotics.com/doc/reference/mesh) node? If so, did you try other formats, e.g., `.obj` which you can easily export from Blender?

ad-daniel, 2021-07-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-07-15 by ad-daniel