cyberbotics/webots #3383: Need more freedom on propeller velocity control.

TaoYibo1866, 2021-07-09:
I am working on a drone project which requires simulation of propeller. When i used setVelocity() to set desired velocity of the motor, i found it took time for the propeller to reach the desired speed, which can not be customized. It took more than 60s for the motor to accelerate from 0 to 900 rad/s. That's too long.

I wish the transient process can be customized. Thank you!

omichel, 2021-07-09:
Did you try to increase the maxTorque of the RotationalMotor in the Propeller?

TaoYibo1866, 2021-07-09:
Thank you for your timely reply, it works. However, I wonder why maxTorque makes a difference, because i didn't set physics on the propeller. I thought there is no dynamics on it.

omichel, 2021-07-09:
The RotationalMotor control uses the maxTorque to generate the target speed even if no physics is enabled.

closed 2021-07-09 by omichel

TaoYibo1866, 2021-07-11:
> The RotationalMotor control uses the maxTorque to generate the target speed even if no physics is enabled.

Sorry for bothering you, where can i find these codes?

mael25, 2021-07-12:
@TaoYibo1866 You can find it there: https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/WbMotor.cpp#L449.

TaoYibo1866, 2021-07-13:
> @TaoYibo1866 You can find it there: https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/WbMotor.cpp#L449.

Thank you! For anyone who has similar question, Motor takes maxTorque (resp. maxForce) as max accelaration value when physics is not enabled.