cyberbotics/webots #3395: Enlarge IndexedLineSet display

ShuffleWire, 2021-07-12:
**Is your feature request related to a problem? Please describe.**
It would be nice if we could specify a width for graphical representation of IndexedLineSet, so that it's more easily visible.
Same thing for joint axis and so forth...
Also, it would be nice, imo, to increase the default "lineScale", as it's too small, and when a user toggle joint axis display on, nothing change, because lines are often too short.
As a side note, changing the line color using a material doesn't seem reliable... (when I remove it, the color don't go back to the default)

**Describe the solution you'd like**
I currently have this : 
[image]
And I wish to be able to do something like this : 
[image]
And maybe use color, but it's not a big deal
The solution would be to add a "width" field in the IndexedLineSet, and the same goes in the WorldInfo, when adding such sort of field next to lineScale would be nice.

**Describe alternatives you've considered**
Using indexed face set and draw big polygon manually, but it's not very handy.

ShuffleWire, 2021-07-19:
Just found out that it's a duplicate of #2512

BenjaminDeleze, 2021-11-15:
Close because duplicate of #2512

closed 2021-11-15 by BenjaminDeleze