cyberbotics/webots #3400: Simulation Server leaking resources

lfelipesv, 2021-07-13:
I am testing the simulation server script and it seems the runWebotsinThread() function (https://github.com/cyberbotics/webots/blob/2f7f45f430746978ecc3dc399fc7dd5d4df362a2/resources/web/server/simulation_server.py#L264) is leaking resources (slowing down the server over time).

I think it might be related to using tornado and the asyncio event loop (https://github.com/cyberbotics/webots/blob/2f7f45f430746978ecc3dc399fc7dd5d4df362a2/resources/web/server/simulation_server.py#L307) inside a thread. Usually tornado does not seem thread safe.

If I comment L310 to L317 it seems to not create the zombie process. But I think it should be a way to close the websocket connection and the loop cleanly.

There are many lines related to closing the client.client_websocket:

https://github.com/cyberbotics/webots/blob/2f7f45f430746978ecc3dc399fc7dd5d4df362a2/resources/web/server/simulation_server.py#L142

But it does not seem to close it effectively.

Regards,

Felipe.

omichel, 2021-07-13:
I believe the main problem is that sometimes, some Webots processes are not quitting and remain active, consuming system resources. Maybe the script could be improved to monitor the active Webots processes (and also Webots controller processes) and kill those which have been inactive for a long period of time.

lfelipesv, 2021-07-13:
I did not have trouble with Webots processes not quitting recently Olivier. Checking the processes leaking resources they are directly associated with the simulation server script.

I think, however, that the problem can be not related to tornado or asyncio. I made some additional tests and it seems that the loop (iter(client.webots_process.stdout.readline, b'')) checking for the webots stdout is still running even after a client is disconnected.

https://github.com/cyberbotics/webots/blob/2f7f45f430746978ecc3dc399fc7dd5d4df362a2/resources/web/server/simulation_server.py#L310

Is there a way to stop this function and complete the thread when a client is disconnected?

The variable client.idle is used for monitoring but I don't understand the meaning of lines 316-317.

lfelipesv, 2021-07-13:
Oh, now thinking better, this termination had to be addressed by killing the webots process https://github.com/cyberbotics/webots/blob/2f7f45f430746978ecc3dc399fc7dd5d4df362a2/resources/web/server/simulation_server.py#L340 right?

omichel, 2021-07-13:
Not sure. Please try and report here any finding you may achieve.

lfelipesv, 2021-07-13:
I added a simple rule right after the blocking iterator 'for' that is used to readlines in https://github.com/cyberbotics/webots/blob/2f7f45f430746978ecc3dc399fc7dd5d4df362a2/resources/web/server/simulation_server.py#L310 

After the modification the snippet it is like:

[image]

In this way, after the webots_process is finished by the kill_webots() function, the thread is completed and the thread is not hanging anymore, freeing the resources.

Does this makes sense Olivier? Because I am not familiar with finishing threads, I don't think it is an elegant solution but it seems to work. What is your opinion on that?

Obs: I tested starting the simulation server and connecting/disconnecting from the streaming viewer. After each connection/disconnection a new process related to the simulation_server.py was created that could be check with the 'htop' command. In a way that after multiple users used the system, we had multiple processes linked with simulation_server.py using the server's memory.

Thank you for your attention!

omichel, 2021-07-13:
Yes, that seems to be a pretty good solution. May I ask you to open a new PR and submit this contribution?

lfelipesv, 2021-07-14:
Sure, I will test it again and submit a PR request with the addition. Thanks!

closed 2021-07-14 by omichel