cyberbotics/webots #3403: Robot controllers written in MATLAB cannot be found/executed if their names contain dashes

jgvictores, 2021-07-13:
**Describe the Bug**

Robot controllers written in MATLAB crash (`Error using eval` and `Unrecognized function or variable '(...)'.` , due to robot controller not found) before execution if their names contain dashes.

I believe to have identified (with the help of @PeterBowman) that it is due to the use of `eval` at: https://github.com/cyberbotics/webots/blob/1b8fdd6243a34c8a3d08a7c1bd84528e9e321609/lib/controller/matlab/launcher.m#L160-L161 which **interprets the `dash` as a `minus`**, and simply truncates the name of the robot controller.

This is especially critical for our projects using https://github.com/cyberbotics/webots-animation-action/blob/cc80c49a0b3fa8f25441b9d93a88ac2c98458181/wb_animation_action/competition.py#L51-L52 which prepends the GitHub user/organization and repository names, which frequently contain dashes.

**Steps to Reproduce**

1. On a PC that has MATLAB, create a robot controller with a dash in the name (e.g. `my-cool-controller.m`
2. Run and see `Error using eval`, etc errors similar to screenshots, below.

**Expected behavior**

Robot controllers written in MATLAB with dashes in their names should be able to run, just as they currently do work for other languages.

Note that a hack for our use case would be introducing a small regex operation (e.g. replace dashes by underscores) at the webots-animation-action level. However, as it seems this issue goes deeper, I would prefer this issue to be solved at the Webots level.

**Screenshots**

- Example using https://github.com/roboticslab-uc3m/iros2021challenge_controller_m (note dash in `roboticslab-uc3m` GitHub organization name).

[image]

- Example illustrating how `eval` tends to truncate at a dash (an attempt at forcing a literal via `\` also failed).
[image]

**System**
- Webots Version: 2021a.
- Operating System: Tested in Windows 10 and Linux Ubuntu 18.04/20.04 with MATLAB 2021a.
- Graphics Card: n/a

omichel, 2021-07-13:
I see two possible solutions to this problem:
1. [Read](https://ch.mathworks.com/help/matlab/ref/fileread.html) the `WEBOTS_CONTROLLER_NAME` file into a string and then `eval()` this string.
2. [Rename](https://ch.mathworks.com/help/matlab/ref/movefile.html) the `WEBOTS_CONTROLLER_NAME` to a temporary file that doesn't contain any hyphen and `eval()` it.

May I let you investigate them and propose a patch in a new PR here?

omichel, 2021-07-13:
Solution 1 seems to be the simplest:
Replace https://github.com/cyberbotics/webots/blob/1b8fdd6243a34c8a3d08a7c1bd84528e9e321609/lib/controller/matlab/launcher.m#L161 with `eval(fileread(WEBOTS_CONTROLLER_NAME))`
Does it works for you?

PeterBowman, 2021-07-13:
While `eval('test-test')` fails for the above reasons, I noticed that `feval('test-test')` on the other hand does not interpret the dash as a minus sign.

omichel, 2021-07-13:
> 
> 
> While `eval('test-test')` fails for the above reasons, I noticed that `feval('test-test')` on the other hand does not interpret the dash as a minus sign.

That's a very good option indeed. Did you try to replace `eval(CONTROLLER_NAME)` with `feval(CONTROLLER_NAME)` in `launcher.m`?

omichel, 2021-07-13:
Last hope is the last option:
- [Rename](https://ch.mathworks.com/help/matlab/ref/movefile.html) the `WEBOTS_CONTROLLER_NAME` to a temporary file that doesn't contain any hyphen and `eval()` it.

PeterBowman, 2021-07-13:
Not sure if a move is the best option, perhaps the original should be preserved? Otherwise use `movefile` instead of `copyfile`:

```matlab
if ~isvarname(WEBOTS_CONTROLLER_NAME)
  newname = matlab.lang.makeValidName(WEBOTS_CONTROLLER_NAME);
  copyfile(append(WEBOTS_CONTROLLER_NAME, '.m'), append(newname, '.m'), 'f');
  eval(newname);
else
  eval(WEBOTS_CONTROLLER_NAME);
end
```

This snippet would replace the following line:

https://github.com/cyberbotics/webots/blob/1b8fdd6243a34c8a3d08a7c1bd84528e9e321609/lib/controller/matlab/launcher.m#L161

jgvictores, 2021-07-13:
Yes, https://github.com/cyberbotics/webots/issues/3403#issuecomment-879003646 via `makeValidName` works! :tada::tada::tada:

PS: Copying does seem safer than moving.

omichel, 2021-07-13:
> PS: Copying does seem safer than moving.

Yes, however, it may be good also to cleanup-up things afterwards, e.g., after `eval(newname);` it would be nice to add something like: `delete append(newname, '.m');`

Could you please open a new PR to propose your contribution? I'll be happy to test it in depth and review it.

closed 2021-07-13 by omichel