cyberbotics/webots #3409: Python `common` module not found on Ubuntu 18.04 running extern controller

stefaniapedrazzi, 2021-07-14:
Python `common` module (newly added in #3387) is not found when running the `slave` controller from the Python example as extern controller on Ubuntu 18.04.

To setup the environment the instructions described in the Webots Guide are followed.
https://www.cyberbotics.com/doc/guide/running-extern-robot-controllers?version=master&tab-os=linux&tab-language=python

However, the controller works correctly if it is run directly by Webots.
So it seems that the instructions to setup the environment for Python extern controllers is missing something.

omichel, 2021-07-14:
That is normal, environment variables for extern controller should be set manually. In this case, it is expected that the `common` module is not found if the `PYTHONPATH` doesn't point to it. To me it is obvious. Maybe we could add in the guide that the `runtime.ini` file is ignored in case of extern controllers and thus the setting of environment variables it contains should be performed manually.

stefaniapedrazzi, 2021-07-14:
Ok. I see now what is this new common module is just a dummy module to show how to use a `runtime.ini`.

stefaniapedrazzi, 2021-07-14:
Yes, it should be explicitly added in the documentation that what is in the runtime.ini should also be set manually.

PR #3413 (2021-07-14): Added explanation about extern controllers and runtime.ini file
Fixes #3409.
closed 2021-07-14 by omichel