cyberbotics/webots #3411: Pen drawing on Mac doesn't evaporates

BenjaminDeleze, 2021-07-14:
**Describe the Bug**
The first segment of line never evaporates in pen.wbt.

**Steps to Reproduce**
1. Go to 'devices/pen.wbt'
2. Run
3. See that the first segment (black) does not evaporate.

**Expected behavior**
The first segment should evaporate as the other ones

**Screenshots**
[image]

**Additional context**
Tested on MacOs big sur
Could not reproduce on Ubuntu 20

ad-daniel, 2021-07-14:
[image]

Same behavior on windows, it gets lighter at the beginning but never actually vanishes.

BenjaminDeleze, 2021-07-14:
The issue is due to a too small ink_intensity (0.000008).
It is not reproducible on Ubuntu in pen.wbt as it does not return the same rand() suite, but if I force the ink_intensity to be that small I can reproduce the issue

closed 2021-07-14 by ad-daniel