cyberbotics/webots #3414: Robot window not working with extern Python controller on Ubuntu 18.04

stefaniapedrazzi, 2021-07-14:
The robot window is not working for robots whose controller is an extern Python controller.

One issue is that on the Webots User Guide it is stated that it is not necessary to set the `LD_LIBRARY_PATH` for running Python extern controller.
However, on Ubuntu 18.04, if the `LD_LIBRARY_PATH` is not set I get these errors in the terminal when starting for example the `slave` controller:

```
stefi@stefi-VirtualBox:~/Desktop/python2/controllers/slave$ python3 -u slave.py 
Error: libgeneric_robot_window.so: cannot open shared object file: No such file or directory (dynamic library)
Error: failed to load /usr/local/webots/resources/projects/plugins/robot_windows/generic/libgeneric.so library
This module is common to both driver and slave controllers (called from slave).
```

After setting the `LD_LIBRARY_PATH` the errors disapper, but still the robot window doesn't work.

omichel, 2021-07-14:
I can reproduce the same problem on Windows, but after adding `$WEBOTS_HOME/lib/controller` to the `PATH` environment variable, everything works, including the robot window.

stefaniapedrazzi, 2021-07-14:
I'm testing on a VM and in case of extern controller the robot window opens but it is not populated.
But in the VM I also have other issues with robot windows and webgl.
I will also test on a native ubuntu 18.04 to compare the results and check if the remaining issues is only specific to VM.

stefaniapedrazzi, 2021-07-14:
I confirm it works correctly for me as well on a native system after setting the LD_LIBRARY_PATH.
I will update the documentation.

lukicdarkoo, 2021-07-14:
[image]

I cannot reproduce on Ubuntu 20.04, I am using Webots development version though

stefaniapedrazzi, 2021-07-14:
You should at least try not in the development system but with webots installed from tarball package and making sure that no specific Webots environmental variables are set.
With this procedure I can reproduce it even on my development system.

lukicdarkoo, 2021-07-14:
> installed from tarball package

With the tarball package it shows the error

closed 2021-07-14 by omichel