cyberbotics/webots #3419: Assets downloaded when copying a robot

stefaniapedrazzi, 2021-07-14:
Blocked assets download dialog displayed is displayed on robot copy.

1. Open 'coupled_motors.wbt'
2. Copy the `GRIPPER` Robot
3. The loading world dialog opens saying it is downloading the assets, moreover it takes really long and don't even complete.

**Expected behavior**
If the robot is already loaded, there should not be any need to download again the assets on copy.

stefaniapedrazzi, 2021-07-14:
Other robot copy give the same issue, for example copying the helicopter robot in `propeller.wbt`.

BenjaminDeleze, 2021-07-14:
I made some tests and it seems that not all robots suffer from this issues.
From what I tested it happens only to robots that are not defined in a PROTO.
However it doesn't seems limited to robots, some solid seems to have this issue too (if you convert RubberDuck to base node for example)
Moreover, talking a PROTO a trying to convert it to base node will also raise this issue.

PR #3428 (2021-07-15): fix assets downloads on copy
**Description**
When copying a object (non-Proto) with a DEF and a USE with urls inside, there was an infinite downloading popup.

This is because a node was deleted before it could finish checking that the URLs were in the cache.
When deleting a node with a WbDownloader, it is not enough to remove the downloads, we also need to decrement the number of total downloads.

Fix #3419
closed 2021-07-15 by BenjaminDeleze