cyberbotics/webots #3420: Robot window doesn't update after restarting an external controller

lukicdarkoo, 2021-07-14:
**Steps to Reproduce**
1. Open `projects/languages/python/worlds/example.wbt`
2. Change controller to `<extern>`
3. Open the robot window
4. Run the `slave.py` extern controller (see the robot window getting the update)
5. Kill and start the extern controller (no update robot window updates)

**System**
 - Operating System: Linux Ubuntu 20.04

stefaniapedrazzi, 2022-10-10:
This issue is still occurring in R2022b.
After killing and restarting the extern controller the robot window is not automatically updated.
The user has to reload the robot window page to make it work again.

stefaniapedrazzi, 2022-10-10:
We discussed that the best solution is probably to close the robot window when the extern controller disconnects.

PR #3544 (2021-08-09): Fix updating robot window on extern controller restart
Fix #3420:
properly notify `WbRobot` instance when an extern controller is restarted.

Remove the unused signal `WbRobot::controllerRestarted`.
closed 2022-10-26 by stefaniapedrazzi