cyberbotics/webots #3421: Wrong version warning on import

stefaniapedrazzi, 2021-07-14:
When exporting/importing a VRML file the following warning message is printed in the console:
```
WARNING: '/home/stefi/Documents/humanoid_marathon.wrl': This file was created with Webots 2.0.0 while you are using Webots R2021b. You may need to adjust urls for textures and meshes, see details in the change log of Webots R2021b.
```

However, the message is wrong because the original file has version VRML v2.0 and not Webots 2.0.

omichel, 2021-07-14:
I believe we should drop this VRML import which is not a useful feature any more since it is way better to refer to external meshes rather than importing VRML files. VRML export could be dropped as well since it is not useful either: I see no use case for that.

stefaniapedrazzi, 2021-07-14:
I don't if any user is using this functionality and probably as you write can be removed.
But with external meshes you can only import one mesh at a time.
This functionality lets you import a whole environment with a single click and creates independent nodes that can then be modified.

I think that is not really replaced by external meshes but it does something differet.

Also note that the functionality is not buggy and works as expected.
There is only a small issue is the warning message.

omichel, 2021-07-14:
I don't see any serious use case for VRML import/export.
I believe that web animations export make VRML export obsolete and Mesh support make import obsolete, even though you have to import meshes one-by-one.
Moreover VRML97 is a very old format, not very relevant any more today.

stefaniapedrazzi, 2022-05-24:
VRML import and export removed in  #4285.

closed 2022-05-24 by stefaniapedrazzi