cyberbotics/webots #3422: ROS test: compilation failure due to `moveit_ros_planning_interface`

stefaniapedrazzi, 2021-07-14:
`moveit_ros_planning_interface` is not automatically installed and `catkin_make` on Ubuntu 18.04 / melodic fails because of it.
I tested with a local copy of the `webots_ros` package, but the error is likely to occur also with the published package.

If this package cannot be installed automatically and needs to be installed manually, then we have to update the ROS tutorial accordingly:
https://cyberbotics.com/doc/guide/tutorial-8-using-ros?version=R2021b

lukicdarkoo, 2021-07-15:
Did you run `rosdep install --from-paths src --ignore-src -r -y`?

stefaniapedrazzi, 2021-07-15:
This issue is solved in #3433 /  #3426.

As written in the description, the documentation should provide all the required instructions to compile and run the `webots_ros` package and it was not the case before.

PR #3433 (2021-07-15): Add the MoveIt dependency to documentation
Fixes #3422
closed 2021-07-15 by lukicdarkoo