cyberbotics/webots #3425: ROS panoramic_view_recorder.wbt is jerky

stefaniapedrazzi, 2021-07-14:
The viewpoint move in the `panoramic_view_recorder.wbt` ROS sample world is jerky and not smooth (as it was supposed to be).
I get this on a VM Ubuntu 18.04 with ROS melodic.

But I will also test on a native system.

stefaniapedrazzi, 2021-07-14:
It seems there are issues also with the `pioneer3at.wbt` simulation because the robot doesn't move at all.

lukicdarkoo, 2021-07-14:
> the robot doesn't move at all 

It is not supposed to move, one should use the `teleop_keyboard` to move the robot.

> prints these errors

I will check those

stefaniapedrazzi, 2021-07-14:
> It is not supposed to move, one should use the teleop_keyboard to move the robot. Tutorials should be updated as soon as we release the new version.

Ok. Thank you. This was not the case before where the robot was moving automatically.

lukicdarkoo, 2021-07-14:
> ```
> [ERROR] [1626276905.451518858]: Could not find parameter robot_description on parameter server
> [example_state_publisher-3] process has died [pid 25906, exit code 1, cmd /opt/ros/melodic/lib/robot_state_publisher/robot_state_publisher __name:=example_state_publisher __log:=/home/stefi/.ros/log/0380e6d0-e4b6-11eb-8987-080027369a75/example_state_publisher-3.log].
> log file: /home/stefi/.ros/log/0380e6d0-e4b6-11eb-8987-080027369a75/example_state_publisher-3*.log
> ```

I couldn't find a way around those. The `ros` controller needs some time to make the `robot_description` parameter available. Therefore, we just respawn the robot_state_publisher until the parameter is ready. Typically, the robot_description comes directly from the URDF file.

stefaniapedrazzi, 2021-07-15:
i confirm I have the same jerky behavior also on native system.

BenjaminDeleze, 2021-07-15:
I also got issues with some ros samples, tested on ubuntu 20 with noetic

Complete_test:
```
[ERROR] [1626356025.611843814]: Client [/complete_test] wants topic /my_robot/camera/recognition_objects to have datatype/md5sum [webots_ros/RecognitionObject/642183f1eab2edbfa9c2ae808be707c2], but our version has [webots_ros/RecognitionObjects/ac0ec54e563936d28b7dec5cf26184c3]. Dropping connection.
```

Pioneer3at:
```
WARNING: ros: The process crashed some time after starting successfully.
Error: wb_lidar_get_layer_range_image() called for a disabled device! Please use: wb_lidar_enable().
WARNING: 'ros' controller crashed.
```

tiago:

```
roslaunch webots_ros tiago.launch
RLException: [tiago.launch] is neither a launch file in package [webots_ros] nor is [webots_ros] a launch file name
The traceback for the exception was written to the log file

```

lukicdarkoo, 2021-07-16:
@BenjaminDeleze It looks like you use the old webots_ros available on apt. R2021b targets the webots_ros package on the develop branch.

BenjaminDeleze, 2021-07-16:
You are right. It solves my errors

stefaniapedrazzi, 2021-07-28:
The problem with the jerky `panoramic_view_recorder` simulation is that the orientation and position changes are not applied in the same simulation step. Even if the source file is explicitly calling the time step callback only after updating both position and orientation, the `Ros::run` loop advances the time in between.

I quickly inspected the changes and the original behavior has been broken by this PR:
https://github.com/cyberbotics/webots/pull/3085

To make this example work correctly we now need to add the `--synchronize` controller argument.

closed 2021-07-28 by stefaniapedrazzi