cyberbotics/webots #3429: getNodeType() return values changed

Thomas-Feldmeier, 2021-07-14:
in /usr/local/webots/include/controller/c/webots/nodes.h (or cpp/webots/Nodes.hpp)

typedef enum {
...
      ACCELEROMETER,
      ALTIMETER,
      BRAKE,
      CAMERA,
...

altimeter is added new to this list, the value of BRAKE is one more than before etc.

i had an external controller, compiled with an old webots. webots::Node::LIDAR was 45, but is now 46. the controller could not find the lidar, because the constant changed. rebuilding the external controller fixed the bug.

move ALTIMETER to the end of the list to prevent change of all the other older constants. or ALTIMETER=xxx, BRAKE=yyy etc.

stefaniapedrazzi, 2021-07-15:
You should always recompile your controller every time you switch to a new major release because the API will probably change and backward compatibility is not guaranteed.
Independently from the changed value of node type there will be other conflicts that won't make the old controller work.

So, it is correct to add new nodes in alphabetical order and not at the end of the list.

stefaniapedrazzi, 2021-07-15:
I will close this issue because this is the expected behavior.
Please reopen it if you want to further discuss about this topic.

Thomas-Feldmeier, 2021-07-16:
Backwards compatibility is important. I have big systems with lots of external controllers. Other people can use it. They may have a different webots version (including nightly build, to fix some errors), and may or may not compile the controllers. I can make most of my controllers run fine with draft webots2021b or nightly build or webots2021a. But getNodeType() is a big problem. Each time an device is added, the enum constant's change. I don't know when this will happen in future. Most API functions don't change over time, but getNodeType() does it periodically.

there is a easy way to make it backward compatible:

include/controllers/c/webots/nodes.h:
...
typedef enum {
...
  WB_NODE_ACCELEROMETER = 35,
  WB_NODE_ALTIMETER = 89, // names are alphabetical ordered, but numbers are ordered by time of adding that device
  WB_NODE_BRAKE = 36, 
...

omichel, 2021-07-16:
I agree backwards compatibility is important and we try our best to maintain backwards compatibility for the source code of controllers. However, maintaining binary backwards compatibility is not possible if we want to continue to improve Webots over time and add new features. Your suggestion is nice, however, we won't implement it as it would be misleading and let people believe we maintain binary backwards compatibility which is wrong. Implementing your proposed change would indeed fix the problem you reported. However, it would encourage people to not recompile their controllers and they would very likely hit others bugs due to the lack of binary compatibility. That would make debugging more difficult for these people.

Ideally, we should even add a version check preventing from running controllers compiled with the wrong version of libController (as it is the case for many common libraries).

closed 2021-07-26 by omichel