cyberbotics/webots #3440: Add a "locking" fonctionnality to brake

ShuffleWire, 2021-07-15:
**Is your feature request related to a problem? Please describe.**
When working on a moving robot with an arm, often the robot could move a long time without using the arm (for instance to bring a part from a room to another). During this movement, the mechanic on the robot doesn't allows any movement of the arm (like an hydraulic actuator, or a locker).
I'm controlling the arm with speed control, but setting the speed to 0 doesn't forbid to drift from it position, especially if the arm is heavy, or if the robot move fast.
Switching to position control doesn't seems to be the good solution, because it will cause the arm the still move a bit, will consume energy if the robot simulate a battery, required a position sensor to get the current position to "lock on", and it's computationally expensive for something that simple...
Using Brake doesn't do the trick, because it add damping, but don't forbid slow drift from a position.

**Describe the solution you'd like**
I want to be able to temporarily lock to part relatively one to the other. I don't want any movement, whatever the force could be.
In term of ODE engine, it would be like an implicit solid merging, which could moreover save computation time.
I could be an option on the join, or maybe on the Brake device, as if we were using an "handbrake" in a car.
However, i don't know if it's possible to (un)merge solid dynamically at ODE level. but it could be implemented latter.

**Describe alternatives you've considered**
For the moment I'm using two connector, on both part I want to lock, and (un)lock the connector when necessary (no snap, no position limitation...). it work well, but add extra complexity I want to get ride of.
I don't know how the Connector work with ODE, or if it's Webots which handle it, but it would be a way to implement something nice, imo.

omichel, 2021-07-15:
Did you try to increase the maxTorque/maxForce values of the motors in your arm? That should help the arm keeping its position while being moved.

ShuffleWire, 2021-07-15:
It's not, actually.
First, I want to keep the max force i've set, because it's usefull otherwise, to detect if the arm could actually make some move etc...
And even if I set dynamically the force to a very high value when it should not move, It won't fix the issue :
An arm will be pull by the gravity, for instance, and because I'm controlling it with speed control, the following will occurs : 
t=0 : arm in position
t = 1*dt : gravity pull the arm, thus, the speed of the arm increase a bit
t = 2*dt : speed control come into play, to bring the velocity to 0, using max force. Note that the velocity is brought back to 0 (which is what the regulation is looking for), not to an opposite speed to bring the arm in place.
t = 3*dt : gravity pull, and so on...
each small displacement after each  small displacement, the arm drift from it position.
Did I get things clear ?
As a side note, i could not risk to get a arm moving even a bit, because I'm using safety system, which, if they detect movement when it shouldn't have any (on the arm), could stop the robot.

omichel, 2021-07-15:
That is strange as we have mobile manipulators, like the KUKA robot which doesn't seems to be affected by what you describe. Could you compare your system with the KUKA youbot example and see where does the difference in behavior come from?

ShuffleWire, 2021-07-15:
Ok, I will have a closer look. But first, my arm is very heavy (300kg empty, up to 1500kg) and that could make thing difficult for Webots. i've not played enough with ERP parameters, maybe it could help...

ShuffleWire, 2021-07-15:
BTW : side question, and mybe could take part of a new issue : speaking of ERP, why ContactProperties use "softERP" ? What is the difference with "ERP" ?

omichel, 2021-07-15:
ERP is global whereas softERP is local.

ShuffleWire, 2021-07-15:
> ERP is global whereas softERP is local.

Very well, but should we change "soft" to "local" it's rather misleading, imo ? 
or do we even actually need to differentiate them ? When defined, it's inside a ContactProperties, already...
But it would break compatibility, though... Just a thought passing by ;)

omichel, 2021-07-15:
> Very well, but should we change "soft" to "local" it's rather misleading, imo ?

No, it is called "softERP" in ODE and we don't want to confuse users with a different naming. ODE has both "ERP" and "softERP" properties, see https://ode.org/ode-latest-userguide.html

closed 2021-11-15 by BenjaminDeleze