cyberbotics/webots #3447: Robot window doesn't work in a virtual machine

lukicdarkoo, 2021-07-15:
**Steps to Reproduce**
1. Open the Python example
2. Double click on the robot
3. Shows the error `[javascript] Uncaught Error: Unable to initialize WebGL. Your browser or machine may not support it. (canvas.js?305_0:26)` and the robot window also doesn't work

**System**
```
$ webots --sysinfo
System: Linux 5.4.0-77-generic x86_64
Processor: AMD Ryzen 7 1800X Eight-Core Processor
Number of cores: 8
OpenGL vendor: VMware, Inc.
OpenGL renderer: llvmpipe (LLVM 10.0.0, 128 bits)
OpenGL version: 4.0 (Compatibility Profile) Mesa 20.0.8
```

ad-daniel, 2021-07-16:
I tried on a nightly build (snap) version on ubuntu 18 and it opens fine

```
System: Linux 5.4.0-42-generic x86_64
Processor: Intel(R) Core(TM) i7-6700 CPU @ 3.40GHz
Number of cores: 4
OpenGL vendor: VMware, Inc.
OpenGL renderer: llvmpipe (LLVM 10.0.0, 256 bits)
OpenGL version: 3.1 Mesa 20.0.8
```

omichel, 2021-07-16:
It works on my VM as well (running on a Windows 10 host):
```
System: Linux 5.8.0-59-generic x86_64
Processor: AMD Ryzen 7 4800HS with Radeon Graphics
Number of cores: 16
OpenGL vendor: VMware, Inc.
OpenGL renderer: SVGA3D; build: RELEASE;  LLVM;
OpenGL version: 4.1 (Compatibility Profile) Mesa 20.2.6
```

lukicdarkoo, 2021-07-16:
> I tried on a nightly build (snap) version on ubuntu 18 virtual machine and it opened fine
> 
> ```
> System: Linux 5.4.0-42-generic x86_64
> Processor: Intel(R) Core(TM) i7-6700 CPU @ 3.40GHz
> Number of cores: 4
> OpenGL vendor: VMware, Inc.
> OpenGL renderer: llvmpipe (LLVM 10.0.0, 256 bits)
> OpenGL version: 3.1 Mesa 20.0.8
> ```

I have the same configuration, Ubuntu 18.04, VirtualBox, and Webots snap. I have just checked again and I am getting the same problem. Does the image in the robot window gets updated?

ad-daniel, 2021-07-16:
No you're right, the robot window does open and is populated but when you actually run the simulation I get the error you mentioned.

omichel, 2021-07-16:
I am not getting the error, but I am running it from the development environment (not from the snap).

BenjaminDeleze, 2021-07-16:
I can reproduce it with the tarball on a VM with Ubuntu 18.

I have this error in this terminal: `ERROR:context_group.cc(145) ContextResult::kFatalFailure: WebGL1 blacklisted`

same config as Daniel

ad-daniel, 2021-07-16:
For me, even opening `Help > openGL information` makes webots crash. `webots --sysinfo` doesn't.

BenjaminDeleze, 2021-07-16:
It is strange, because I could stream a simulation from inside the VM and play it in a browser in the VM as well. So it can create a WebGL context (at least in browsers).

ad-daniel, 2021-07-16:
Tested the same setup with `R2021a` snap and it works. I don't know how much about it was changed this release, but it might be a regression.

stefaniapedrazzi, 2021-07-26:
> Tested the same setup with R2021a snap and it works. I don't know how much about it was changed this release, but it might be a regression.

Robot windows in R2021a are not using WebGL.

The exact same code to get the WebGL context in Firefox works but not in if the robot window is started by Webots (on a Ubuntu 18.04 VM).

stefaniapedrazzi, 2021-07-26:
It seems this is a Qt bug of the `QtWebEngine` module.
I tried to revert to Qt 5.15.0 (see #3089) and the robot windows work correctly in the VM.

stefaniapedrazzi, 2021-08-02:
Robot windows in VM also work correctly with a standard installation of Qt 5.15.1 (for testing I didn't recompile libqjpeg and QWebSockets).
I will try to identify the changes and write a bug report.

stefaniapedrazzi, 2021-08-02:
Here is the link to the bug report:
https://bugreports.qt.io/browse/QTBUG-95536

omichel, 2022-06-13:
Since we moved the robot in windows to the web browser, this issue is not relevant any more.

closed 2022-06-13 by omichel