cyberbotics/webots #3448: Wrong urls in snap

BenjaminDeleze, 2021-07-15:
The urls should have the form "https://raw.githubusercontent.com/.../R2021b/..."
But the snap package have urls with the commit number: "https://raw.githubusercontent.com/.../R2021b/..."

All the other packages (.exe, .dmg, .deb, .tar.gz) are correct.

I think the snap is wrong because he is created separately from the other one.

BenjaminDeleze, 2021-07-16:
It seems that I was right, because the snap of the nightly build has correct urls: "https://raw.githubusercontent.com/.../R2021b/..." 
It is correct since the HEAD is still on the R2021b tag.

lukicdarkoo, 2021-07-16:
> I think it is because of the order in which we created the packet today:

Yesterday, I re-ran the snap package creation just after all packages got built (I had different problems because of the same reason)

BenjaminDeleze, 2021-07-16:
Okay, that is strange. So I don't know why it is correct in the nightly but not in the release

BenjaminDeleze, 2021-07-16:
I tested again and it is fixed. I think I tested yesterday with the first snap package (before it was recreated)

closed 2021-07-16 by BenjaminDeleze