cyberbotics/webots #3450: remote urls in stub files which get replaced on download

Thomas-Feldmeier, 2021-07-16:
e.g.
change /usr/local/webots/projectds/default/worlds/textures/wood.jpg to:
https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/default/worlds/textures/wood.jpg

sample world motor: SolidBox > appearance > baseColorMap > url
"textures/wood.jpg"

webots loads the sample world motor, wants to load textures/wood.jpg, finds the file in above location, sees it is only a stub to webots github, downloads the stub link and replaces the stub with the downloaded file before displaying it.
the next time the world is loaded, the texture is already there.

this way
* webots users don't see any changes (compatible)
* webots installation download is fast
* webots downloads additional resources only once

omichel, 2021-07-16:
Unfortunately, this doesn't work because the file for example `/usr/local/webots/projectds/default/worlds/textures/wood.jpg` doesn't exist and hence we cannot infer the corresponding 
https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/default/worlds/textures/wood.jpg unless we create a dictionary for all previous textures and meshes used in previous versions, which would be cumbersome.
Moreover, we want to encourage users to upgrade their simulations and proto files to the new URL system which is more scalable than the previous one.

Thomas-Feldmeier, 2021-07-16:
A script could do all the work: in project folder, search for strings "https://raw.github...", save this list (temporary "dictionary")
replace these strings by relative path to the stub file
create all stub files from the list

how is the new url system working? where are the files cached? how long? is it possible to change these settings?

the old system run without internet connection required, even offline documentation was included. is this planned to be possible again?

the url links have webots version numbers inside. over the years, user made worlds/protos will have links to old versions. must the user update all of these? what happens, if both webots/R2021b/.../wood.jpg and webots/R2022a.../wood.jpg is used? maybe file duplication, on server side and on client side (cache). this problem gets bigger with time.

encourage users to use URL system? over the years, some internet links will vanish. old content will be lost then.

if webots finds a world/proto with an old version, it is now printing an warning message about new url system, even if it is an .proto with nothing inside but group {}. users are now forced to change the webots version number in all selfmade worlds and protos just to get rid of that warning?

with time, old content will grow and get more important. the new url system will make problems in the future.

Thomas-Feldmeier, 2021-07-19:
empty stub files with zero length would be more easy/efficient

on webots deinstallation, does the new url system clear the file cache to prevent resource leak?

Thomas-Feldmeier, 2021-07-19:
linux only:
to change all .proto or .wbt files in current folder to R2021b version (to get rid of that pesky warning)

find . -type f -exec sed -i "s/R2021a/R2021b/g" {} \;
find . -type f -exec sed -i "s/R2020b/R2021b/g" {} \;

omichel, 2021-07-26:
> on webots deinstallation, does the new url system clear the file cache to prevent resource leak?

No, you have to clear it from the Webots preferences.

closed 2021-11-15 by BenjaminDeleze