cyberbotics/webots #3462: Python virtual environments don't work with R2021b

dfremont, 2021-07-16:
**Describe the Bug**
Python virtual environments no longer work with Webots R2021b. With R2021a (and earlier versions - I think it worked going back to R2018) I could launch Webots inside a virtualenv and my controllers would be able to use modules installed in that virtualenv. Under R2021b the system Python gets used even if Webots is launched from inside a virtualenv.

(Given that there seems to be nothing in the documentation about virtual environments, it's possible the behavior under R2021a was just an accident and so this isn't a bug. Even in that case, it would be great if you could restore the R2021a behavior since it was extremely convenient - I don't want to have to change the "Python command" option when moving between projects.)

**Steps to Reproduce**
1. In a terminal, enter a virtual environment (I'm using [Poetry](https://python-poetry.org/) so I do `poetry shell`). Install some module there which is not installed in the system Python.
2. Launch Webots from the terminal.
3. Add a Robot node with a Python controller which imports the module.
4. Running the simulation yields an `ImportError` in the console.

**Expected behavior**
In R2021a, the controller script worked correctly, finding the module installed in the virtualenv.

**System**
 - Operating System: macOS Big Sur (11.4)
 - Graphics Card: Intel Iris Plus Graphics 655
 - Python version: 3.7

stefaniapedrazzi, 2021-07-19:
We never officially support Python virtual environments in Webots.
We will check if it is possible to restore this functionality.

dfremont, 2021-07-19:
Great, thank you Stefania!

stefaniapedrazzi, 2021-07-28:
Should be fixed in  #3506.

@dfremont the fix will be available from tomorrow in the [nightly build](https://github.com/cyberbotics/webots/tags) of Webots R2021c.
We will be happy to have a feedback if this is fixing your issue.
Otherwise please feel free to reopen this issue.

closed 2021-07-28 by stefaniapedrazzi