cyberbotics/webots #3468: Distance sensors don't see through transparent objects

BenjaminDeleze, 2021-07-19:
**Describe the Bug**
Distance sensors don't see through fully transparent objects (objects with a transparency set to 1).

It seems harder to fix than #3405 because it is related to ray collisions.

**Steps to Reproduce**
1. Open `laser_pointer.wbt`
2. Change the transparency parameter of the green box to 1
3. The lasers still stop on the box

**Expected behavior**
The lasers should go through the box.

llessieux, 2021-09-29:
New to the code base, but if I read it correctly it should be too difficult to fix it.
Change WbGeometry to have a IsTransparent function.
Modify the SensorRay in WbDistanceSensor to have a function to set a mIgnoreTransparent call that for the laser case.
Then in SetCollision check the bool and check IsTransparent on the incoming geometry and ignore if needed.
I tried it and it seems to work but you need to set the Transparency to 1 on the bounding box too.

llessieux, 2021-09-29:
alternatively adding 
  if (mRayType == LASER  && object->isTransparent()) {
    return;
  }
in WbDistanceSensor::rayCollisionCallback
with the isTransparent function added to WbGeometry will also do it.

omichel, 2021-09-29:
Can you please go ahead with a PR implementing that fix? We would be happy to review it and eventually merge it.

llessieux, 2021-09-29:
Sure, let me get familiar with the branching system etc. I should be able to add that today or tomorrow (given time difference).

omichel, 2021-09-29:
I just sent you a GitHub invite to become a member of the committers group, so that you don't need to fork the repo. You can simply check it out on your local machine, create your own branch, make modifications, push it and create a pull request for it. Let me know if you have any question.

llessieux, 2021-09-29:
Thanks. 
I should be fine now ;)
Long time no see ;) It has been a while since I made the KHR-2HV models  :)

omichel, 2021-09-29:
> It has been a while since I made the KHR-2HV models :)

Yes several years!

closed 2021-11-15 by BenjaminDeleze