cyberbotics/webots #3471: CameraRecognitionObject doesn't work with Python 3.9

sciencectn, 2021-07-19:
**Describe the Bug**
Using most getter methods on the [CameraRecognitionObject](https://cyberbotics.com/doc/reference/camera?tab-language=python#camera-recognition-object) with Python 3.9 results in a segfault. It works fine in Python 3.8. 

**Steps to Reproduce**
1. Open camera_recognition.wbt in $WEBOTS_HOME/projects/samples/devices/worlds/camera_recognition.wbt
2. Open the camera and try using any of the methods on `CameraRecognitionObject`, like `get_id`, `get_position_on_image`. 
Example here
```
"""camera_crash controller."""

from controller import Robot
robot = Robot()

timestep = 1000
camera = robot.getDevice("camera")
camera.enable(timestep)
camera.recognitionEnable(timestep)

while robot.step(timestep) != -1:
    num_objects = camera.getRecognitionNumberOfObjects()
    objs = camera.getRecognitionObjects()
    for obj in objs:
        print("model=",obj.get_model())
        print("id=",obj.get_id())   # it crashes on this line 
```

**System**
 - Operating System: tried both Ubuntu 20.04 and macOS 11.4
 - Graphics Card: RTX 2070 on Ubuntu, AMD Radeon Pro 5500M on the Mac.

stefaniapedrazzi, 2021-07-22:
I can also reproduce this on Ubuntu 18.04.

stefaniapedrazzi, 2021-07-22:
The crash is due to the methods `PyInt_FromLong`, `PyFloat_FromDouble`, etc. that crashes on Python 3.9 (I'm using 3.9.6).

This doesn't seem to bet limited to the `CameraRecognitionObject` functions but should occur with all the occurrences of these functions (`Camera`, `ContactPoint`, `Lidar`, `RangeFinder`).

However, `PyBytes_FromStringAndSize` used works correctly.

stefaniapedrazzi, 2021-07-26:
The bug is due to this change introduced in #3104:
https://github.com/cyberbotics/webots/blob/d5934afadbe124650c747994362c3c70ba2f218c/src/controller/python/controller.i#L19

After reverting this line to the following everything still work correctly:
```
%module controller 
```

stefaniapedrazzi, 2021-07-26:
As mentioned in http://swig.10945.n7.nabble.com/How-to-release-Python-GIL-td5027.html#a5029
other than setting the threads it seems that we also need to use the SWIG_PYTHON_THREAD_(BEGIN/END)_(ALLOW/BLOCK) macros when Python C API functions are called.

Here is another threads explaining these macros:
http://swig.10945.n7.nabble.com/Difference-between-SWIG-PYTHON-THREAD-BEGIN-ALLOW-and-SWIG-PYTHON-END-BLOCK-td9942.html

closed 2021-07-26 by stefaniapedrazzi