cyberbotics/webots #3476: Use actual boolean in Lidar::isRotating

ShuffleWire, 2021-07-20:
Currently, Fixed or Rotating type of a Lidar is determined using 
```
bool isRotating() const { return mActualType.startsWith('r', Qt::CaseInsensitive); }
```
However, it's not very efficient as this is called often.
Would not be better to actually store a bool in `mActualType` instead of a `QString`, and only use "Fixed/Rotating" in the display ?

Also, I do think that something like the "Sperical" tick-box could be nice.

stefaniapedrazzi, 2021-07-22:
This is fine, but note that this won't really speedup the simulation as performance gain is really minimal.
For optimizations we should first identify bottlenecks and then try to solve them instead of randomly search in the code.

lukicdarkoo, 2021-07-22:
@ShuffleWire The `WbLidar::updatePointCloud` method is probably the biggest bottleneck (by far) in the LiDAR implementation:
https://github.com/cyberbotics/webots/blob/c5637734a664733a585219c178eb753c0ff8f37b/src/webots/nodes/WbLidar.cpp#L354-L381
If you ever find a time, please try making a PR that optimizes it (see some ideas in https://github.com/cyberbotics/webots/issues/3046).

ShuffleWire, 2021-07-22:
> @ShuffleWire The `WbLidar::updatePointCloud` method is probably the biggest bottleneck (by far) in the LiDAR implementation:
> 
> If you ever find a time, please try making a PR that optimizes it (see some ideas in #3046).

I will see if I can do anything, but I couldn't promise much... I will take a look, still !

closed 2021-07-22 by mael25