cyberbotics/webots #3477: Optimization on Lidar

ShuffleWire, 2021-07-20:
Its seems that rendering camera for a Lidar take into account textures, as all camera, is that correct ?
If so, I do think that textures are not necessary, because Lidar only use the shape (ie depth buffer) of the object.
Hence, we might get some speed improvement by ignoring textures in that case.
Did I got things right ?

omichel, 2021-07-26:
Yes, that makes sense.
Feel free go ahead with the implementation.
I will be happy to review your PR about it and see what performance improvement we can get from it.

llessieux, 2021-10-01:
Is that actually true? 
The WbLidar creates a camera of type l , this leads to
void WbWrenCamera::init() {
  mFirstRenderingCall = true;
  mIsCopyingEnabled = false;
  mIsColor = mType == 'c' || mType == 's';
mIsColor is then false.

then later in makeWrenCurrent
  if (!mIsColor)
    materialName = "encodeDepth";
and that should be the name of the shader used. 

WrShaderProgram *WbWrenShaders::encodeDepthShader() {
....
    ::buildShader(gShaders[SHADER_ENCODE_DEPTH], QFileInfo("gl:shaders/encode_depth.vert"),
                  QFileInfo("gl:shaders/encode_depth.frag"));
  }
It should be using the encode_depth.frag/vert which are not using any texture or coloring at all.

So as far as I can see it seems to be doing the right thing already ;)

omichel, 2021-10-01:
Thank you for exploring this. I didn't check it yet, but yes, what you say makes actually sense. So, texturing doesn't seem to be used indeed, which is fine.

closed 2021-10-01 by omichel