cyberbotics/webots #3488: Standard ROS controller crashing on simulation start when using use-ros-control flag

tiborbs1, 2021-07-22:
**Bug description**
When starting webots simulation with a robot which has a standard ros controller, the ros controller crashes producing the error attached in the screenshot below. The error only occurs when the use-ros-control flag is set. 

**Steps to Reproduce**
1. Start roscore
2. Start webots 
3. Load world with a robot with the standard ros controller 
4. Start the simulation
4. See error

**Screenshots**
[image]

**System**
 - Operating System: Ubuntu 20.04.2 LTS 
 - Webots version: R2021b
 - ROS version: noetic

**Additional context**
The error doesn't always occur on startup. Most of the time it will, but there are times when it will crash a little bit later.

lukicdarkoo, 2021-07-22:
Do you have an idea of how does your use-case differs from e.g. `roslaunch webots_ros pioneer3at.launch`? If I understood (on Discord) the `pioneer3at` example works for you?

tiborbs1, 2021-07-22:
The pioneer3at example doesn't work as well. It worked before, but now it doesn't. I've tried reinstalling webots and rebuilding webots_ros, but it hasn't helped.

lukicdarkoo, 2021-07-23:
> It worked before

With the previous Webots version? Did you compile the webots_ros package from master?

tiborbs1, 2021-07-23:
Yeah, it worked with the previous Webots version. I compiled the webots_ros package from master. I solved the issuse by compiling webots from master. Everything is working now. Before I did that, I tried to install it from the debian package on an another machine. It showed the same error.

lukicdarkoo, 2021-07-23:
It might be related to #3486. We compile the Debian package on Ubuntu 18.04, so all controllers are linked against Ubuntu 18.04 libraries.

Also, using the tarball package should solve the problem (https://github.com/cyberbotics/webots/releases/download/R2021b/webots-R2021b-x86-64.tar.bz2).

tiborbs1, 2021-07-23:
It is probably related because when I installed it I "solved" the linking issue by adding a symbolic link from libboost_system.so-1.65.1 (the missing library) to libboost_system.so-1.67.0.

lukicdarkoo, 2021-08-06:
ROS 1 and R2021b compatibility:
- ROS Noetic works with the tarball for Ubuntu 20.04 (https://github.com/cyberbotics/webots/releases/download/R2021b/webots-R2021b-x86-64.tar.bz2)
- ROS Melodic works with the Debian packages and tarball for Ubuntu 18.04

As of Webots R2021c, ROS Noetic will work with the Debian package (but ROS Melodic will not), see #3507

closed 2021-08-06 by lukicdarkoo