cyberbotics/webots #3489: hiddenField vrmlField deprecatedField undocumented

Thomas-Feldmeier, 2021-07-22:
please add some documentation about "hiddenField" and "vrmField" and maybe "deprecatedField"

/usr/local/webots/project/default/protos/Supervisor.proto:

```
PROTO Supervisor [
  vrmlField SFVec3f    translation         0 0 0
  vrmlField SFRotation rotation            0 1 0 0
  vrmlField SFVec3f    scale               1 1 1
  vrmlField MFNode     children            []
  field     SFString   name                "supervisor"
...
  field     SFString   remoteControl       ""
  hiddenField SFVec3f  linearVelocity      0 0 0
  hiddenField SFVec3f  angularVelocity     0 0 0
  deprecatedField SFString data            ""
]
```
project/humans/pedestrian/protos/Pedestrian.proto:
```
PROTO Pedestrian [
...
  hiddenField SFFloat    rightLegAngle         0.0
  hiddenField SFFloat    rightLowerLegAngle    0.0
...
```

hiddenField is useful: these fields can be read/write by an supervisor, but are invisible in webots scene tree. Pedestrian controller is using these to move limbs without any motor.

what it the difference between vrmlField and field ?

Thomas-Feldmeier, 2021-07-22:
are vrmlField the exposedField from [vrml 2.0 specification](http://gun.teipir.gr/VRML-amgem/spec/part1/nodesRef.html) ?

omichel, 2021-07-26:
No, `vrmlField` indicates a field directly inherited from the VRML97 specification, whereas `field` indicates an extended field, specific to Webots.

omichel, 2021-07-26:
However, we are probably going to remove this distinction soon as we plan to drop references to the outdated VRML97 standard.

closed 2021-11-15 by BenjaminDeleze