cyberbotics/webots #3495: Release to ROS buildfarm

Timple, 2021-07-23:
**Is your feature request related to a problem? Please describe.**
Non-typical installation steps for ROS users.

**Describe the solution you'd like**
Release the debian to the ROS buildfarm.
It would be nice if ROS users can simply `apt-get install webots` instead of configuring a ppa or downloading debians manually.

**Describe alternatives you've considered**
snap install is also possible. But having to no additional steps is nice. Especially for CI purposes which have to do this often and require extra scripts etc.

**Additional notes**
Might lower the bar for new users as well, making adoption of webots even easier (just like one can `apt install gazebo11` already).

lukicdarkoo, 2021-07-23:
We used to release Webots with the `webots_ros2` package, but the ROS buildfarm started complaining as THE package was too big. Since R2021b we significantly reduced the size, so it might be feasible now.

What is your use case exactly? Webots gets automatically installed as soon as you start any simulation from the `webots_ros2` package. Do you need it for CI?

Timple, 2021-07-23:
Personally I'm not to fond of scripts that magically install binaries in locations I don't expect (figured out it was the .ros folder on a colleagues PC). Also it's probably an ever increasing diskspace consumptions as every release gets added.
Note that I do get the 'need' for such a system to easy first-time usage for new users.

CI is something we're investigating now, so that would help. I know you provide dockers, but we also have our own dockers as base image, so that's tricky.

Also it's easier to notice when thinks change when using apt. Our textures broke with the new release. But we never noticed the upgrade as we used the snap package.

So mostly: it's in line with all other packages we use and thus easier on our tooling and colleagues :slightly_smiling_face:

lukicdarkoo, 2021-07-23:
Given the fact it requires only two additional commands to install Webots apt:
```
wget -qO- https://cyberbotics.com/Cyberbotics.asc | sudo apt-key add -
sudo apt-add-repository 'deb https://cyberbotics.com/debian/ binary-amd64/'
```
I am afraid it's hard to justify maintaining one more distribution. The only reason would be a consistency with the other ROS packages.

Timple, 2021-07-23:
Alrighty!

closed 2021-07-23 by Timple