cyberbotics/webots #3497: Is it "SolidReference" BUG

chinahuangyong, 2021-07-23:
when i use the "SolidReference",  but it .....
[image]

chinahuangyong, 2021-07-23:
This is my project
[stdbot_sim.zip](https://github.com/cyberbotics/webots/files/6870368/stdbot_sim.zip)

ShuffleWire, 2021-07-24:
When working on an issue, first try to explain which result you expect "when i use the "SolidReference", but it ....." is not enough as issue ;) 
AFAIK, it should create a joint to close the loop, but it fall of as being not properly anchored
Secondly try to provide the smallest example that show the issue. For of, if the problem is not related to the contoler, remove it. And try to provide the smallest model possible.
I've made that, at least a first version, so others could take a look.
[stdbot_sim.zip](https://github.com/cyberbotics/webots/files/6872955/stdbot_sim.zip)

chinahuangyong, 2021-07-26:
think you, got it ,and this is a success gif

chinahuangyong, 2021-07-26:
[image]

closed 2021-07-26 by chinahuangyong