cyberbotics/webots #3498: Remote resource cannot be downloaded

chenxin199305, 2021-07-23:
After installing the newest version of webots, I find that the texture of the scene cannot be downloaded. As I am in mainland China, I am not sure if it is VPN problem. But when I copy and paste the resource link and open my VPN to download it, it also shows 404. So, I am here writing this issue to please request a re-check from you that all the resources in the remote repository are downloadable. Thanks.

Thomas-Feldmeier, 2021-07-26:
the link you give does not work for me too (has nothing to do with VPN problem). try this one:

https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/thread_metal_plate/oval_thread_metal_plate_base_color.jpg

https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/thread_metal_plate/oval_thread_metal_plate_base_color.jpg

i got this link by adding shape, appearance > Proto nodes > appearance > ThreadMetalPlate, convert to base Nodes, baseColorMap url

TaoYibo1866, 2021-07-28:
I tried to copy those textures from my older version webots to 2021b folder, but didn't help. Does anyone know where are those downloaded files located in 2021b?

stefaniapedrazzi, 2021-07-28:
They are automatically downloaded and loaded by Webots.

If you want to download/copy them manually, then make sure to change the URL of the textures so that they point to your resources and not to the default URL. Because otherwise Webots will continue to try to download them from the web.

Note that Webots default URLs are correct and this seems a problem with your network connection.

TaoYibo1866, 2021-07-28:
> They are automatically downloaded and loaded by Webots.
>
> If you want to download/copy them manually, then make sure to change the URL of the textures so that they point to your resources and not to [...]

Thank you for your reply. As we are in mainland China, we cannot access githubusercontent. We hope webots can give us a link to download all downloadable resources so that we can manually place it in the right place. This will mean a lot to us!

stefaniapedrazzi, 2021-07-28:
> Thank you for your reply. As we are in mainland China, we cannot access githubusercontent. We hope webots can give us a link to download all downloadable resources so that we can manually place it in the right place. This will mean a lot to us!

This is not possible with Webots R2021b.
We are discussing about possible solutions that could be included in the next version but for the moment there is nothing concrete yet.

TaoYibo1866, 2021-07-28:
> This is not possible with Webots R2021b.
> We are discussing about possible solutions that could be included in the next version but for the moment there is nothing concrete yet.

Looking forward to it, thanks!

Thomas-Feldmeier, 2021-07-28:
best thing is to use webots R2021a or wait until webots R2021b revision comes out (i hope this comes)

dirty hack to "download" resources:

webots 2021b, copy webots/project folder to folder B and C (C is backup if you mess up)
webots 2021a, copy webots/project folder to folder A
now copy folder A to B, but don't overwrite already existing files. this should add all missing .png .jpg etc to their old location

now write a little program or a script to find all .wbt or .proto (maybe more) files in folder B, replace like this:
    "https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/road/protos/textures/road_line_triangle.png"
    "textures/road_line_triangle.png"

"https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/OOO/AAA/BBB"
-> "AAA/BBB"

where AAA and BBB does not contain / or "
OOO is the rest, does not contain ", but can contain /

copy folder B to webots/project folder of webots R2021b. maybe you have to give proper rights to that folder

this hack should theoretically work, but i have not used it myself except for mybots sample world where it worked, so maybe there are more problems.

TaoYibo1866, 2021-07-29:
> best thing is to use webots R2021a or wait until webots R2021b revision comes out (i hope this comes)
>
> dirty hack to "download" resources:
>
> webots 2021b, copy webots/project folder to folder B and C [...]

Thank you! I just tried your method and it worked. However, i decided to use webots 2021a at last.

Thomas-Feldmeier, 2021-07-29:
> Thank you! I just tried your method and it worked. However, i decided to use webots 2021a at last.

Can you please share your script/program to find files and replace text? This could help others.

TaoYibo1866, 2021-07-29:
> > Thank you! I just tried your method and it worked. However, i decided to use webots 2021a at last.
> 
> Can you please share your script/program to find files and replace text? This could help others.

I merely rename some of the links manually to get my simulation running, sorry for that.

closed 2021-11-15 by BenjaminDeleze