cyberbotics/webots #3503: darwin-op can't open robot window

Thomas-Feldmeier, 2021-07-26:
sample world darwin-op

double click on robot or right click robot > show robot window:
```
Error: libzip.so.4: cannot open shared object file: No such file or directory (dynamic library)
Error: robot window initialization failed
Error: Cannot load the "/usr/local/webots/projects/robots/robotis/darwin-op/plugins/robot_windows/robotis-op2_window/librobotis-op2_window.so" robot window library.
```

the file is there:

```
ls -al /usr/local/webots/projects/robots/robotis/darwin-op/plugins/robot_windows/robotis-op2_window/librobotis-op2_window.so

-rwxr-xr-x 1 root root 172496 Jul 16 14:28 /usr/local/webots/projects/robots/robotis/darwin-op/plugins/robot_windows/robotis-op2_window/librobotis-op2_window.so

```

stefaniapedrazzi, 2021-07-26:
> webots R2021b; ubuntu 20.04; i don't have a real darwin-op-robot, so a remote control to a real one will fail and that's why there is no robot window?

No, it doesn't seem the error is related on the availability of a real robot.
It seems simply that the `libzip.so.4` library needed to run it is not available on the system.
Which package did you use to install Webots ? Snap, debian or tarball?

stefaniapedrazzi, 2021-07-26:
I tested on a clean machine and it works fine if Webots is installed with the tarball or snap package but I can reproduce the issue with the debian package.

stefaniapedrazzi, 2021-07-26:
On Ubuntu 20.04 the available version libzip is libzip5 and in Ubuntu 18.04 is libzip4.
This makes impossible to distribute a single Debian package fully working out of the box on both Ubuntu distribution.
But we could reintroduce the limitation of the support of the Debian package only for the latest Ubuntu LTS (currently Ubuntu 20.04) as it was the case a couple of years ago.
To run Webots on Ubuntu 18.04 users will have to use the snap or tarball package.

A workaround to use the robotis robot window on Ubuntu 20.04 is to recompile the library:
1) Open any C++ file of the robot window (for example `$WEBOTS/projects/robots/robotis/darwin-op/plugins/robot_windows/robotis-op2_window/entry_points.cpp`)
2) Press the clean button on the text editor toolbar
3) Accept the dialog about project relocation
4) Press the build button on the text editor toolbar
5) Reload the simulation -> the robot window library is now linked with the correct version of `libzip` and can be correctly opened

Note that this is not a regression of Webots R2021b and there was the exact same issue with the previous versions.

omichel, 2021-07-26:
I would suggest to drop Ubuntu 18.04 support for the `.deb` package and support only 20.04. This might also fix the problem we have with ROS Noetic.

Thomas-Feldmeier, 2021-07-26:
webots_2021b_amd64.deb

PR #3507 (2021-07-28): Create Debian package from Ubuntu 20.04
Drop support of Debian package on Ubuntu 18.04.
It is already stated in the documentation that the Debian package works only on the latest Ubuntu LTS:
https://www.cyberbotics.com/doc/guide/installation-procedure#installation-on-linux

This should fix #3503 and #3486.
closed 2021-07-28 by stefaniapedrazzi