cyberbotics/webots #3505: Add version check in libController

omichel, 2021-07-26:
Following-up #3429, it would be nice to add a version check preventing from running controllers compiled with the wrong version of libController (as it is the case for many common libraries). This would certainly avoid some bug reports about misbehaving controllers.
The Webots version should be included in the compiled controllers and checked against the libController version. If the version do not match, an error message should be displayed, asking users to recompile their controller programs with the current version of the libController and the controller should terminate.

Thomas-Feldmeier, 2021-07-26:
there may be closed source with only binary but no source code available (proprietary code etc.). don't you want webots to be part of big commercial solutions?

omichel, 2021-07-27:
No: many open source libraries have version control check, like [openssl](https://www.openssl.org), as executing a binary program linked with the wrong version of a library may cause bugs.

Thomas-Feldmeier, 2021-07-27:
What happens if one program needs openssl 2.0 and another one needs openssl 3.0. Can both run on the same computer?

What happens if one program needs webots 2021a and another one needs webots 2021b. Can both run on the same computer?

omichel, 2021-07-27:
In the case of webots, it is possible to install different versions on the same computer. The controller program will use the version that is pointed to by the LD_LIBRARY_PATH environment variable. This variable is automatically set by Webots for normal controllers and should be manually set for extern controllers. For openssl, I am not sure which mechanism is used, but I am pretty sure it is possible to have different versions co-existing on the same computer.

Thomas-Feldmeier, 2021-07-27:
openssl is secret communication. Here backwards compability is bad: an attacker could use the old insecure standard.

Collaboration needs an common standard to communicate. Here backwards compatibility is good: everyone should be able to work with everyone. The standard opens the market, progress is made.

Let's say Robot A and Robot B are from different manufacturers, but run in the same factory. Both should connect to the same simulation.

omichel, 2021-07-27:
Supporting binary backwards compatibility is a feature that is very costly in terms of software development, maintenance and testing. It is not in our roadmap, unless somebody wants to sponsor it. Note that very few software libraries support this, not only security software.

Thomas-Feldmeier, 2021-07-27:
Binary backwards compatibility is not required.

A controller and webots talk to each other by sending commands/data by a two-directional stream. The data structure and commands must be backwards compatible. As the controller only interacts with webots through this stream, that is the only requirement for backwards compatibility. This can be done without much effort. Just add new commands without removing/changing old ones. If adding a new device, don't change the magic numbers of the old devices.

BenjaminDeleze, 2023-06-13:
Implemented in #5896

PR #5896 (2023-02-17): Add version comparison between libController and Webots
Resolves #3505.

With extern controllers (namely TCP ones) and the `webots_ros2` package it happens that the user uses different versions of Webots and of the libController, which is forbidden, as both could be incompatible with each other.

Webots is now sending its version in the configure message of the robot. The version is compared on the libController side and a warning is displayed if they are not equal.
closed 2023-06-13 by BenjaminDeleze