cyberbotics/webots #3508: Online simulation connection error

yuchen93, 2021-07-27:
**Describe the Bug**
I was able to run online simulation on my own server previously but I started to getting a lot of connection error more recently and now the simulation won't run anymore. I get 'connection closed abnormally' after it tries to establish connections. On chrome, the error code is 1006 but on firefox the error code is 1015. I tried updating/renewing ssl certificates but that didn't help. There is no code change in between so I am very confused. Please help me understand what the possible issue could be. 

Thanks.

**System**
 - Operating System: Ubuntu (AWS EC2 G3 server)

omichel, 2021-07-27:
It might be related to this [bug](https://bugreports.qt.io/browse/QTBUG-57665). Did you compiled Qt with this [patch](https://github.com/cyberbotics/webots/wiki/Qt-compilation#patch-and-recompile-qwebsockets)?

stefaniapedrazzi, 2021-07-28:
> It might be related to this [bug](https://bugreports.qt.io/browse/QTBUG-57665). Did you compiled Qt with this [patch](https://github.com/cyberbotics/webots/wiki/Qt-compilation#patch-and-recompile-qwebsockets)?

@ad-daniel could you confirm if you applied the patch when creating the latest Qt 5.15.2 package?

ad-daniel, 2021-07-28:
I followed the steps of the wiki so I believe I did yes.
@yuchen93 Which version of Webots are you using? Did you upgrade recently to 2021b and the problems started after that?

yuchen93, 2021-08-03:
It happened before I upgrade, and I thought upgrading might fix it but it didn't. Everything was working before, so I am a bit confused about what could have changed.

yuchen93, 2021-08-03:
Another thing I observe is that the session server often complains the simulation server is not responding. I need to restart several times to get them to both work and then I get the 'connection closed abnormally error' whenever trying to establish a remote simulation.

closed 2021-11-15 by BenjaminDeleze