cyberbotics/webots #3523: mouse shift+move wrong direction if within rotation

Thomas-Feldmeier, 2021-07-29:
world with coordinate system ENU (maybe not required, have not tested it)

add a transform with a rotation, i used this one but i think any would do:
```
#VRML_SIM R2021b utf8
# rotates things which were build with NUE coordinate system (UP=y coordinate)
# to the ENU coordinate system (UP=z coordinate) if needed

PROTO ConvertCoord [
  field MFNode      children  []
]
{
  Transform {
%{ if context.coordinate_system == "ENU" then }%
    rotation 1 0 0 1.5708
%{ end }%
    children IS children
  }
}
```

now add some solid to the children of that transform (e.g. PottedTree)
press shift key and click on the solid to and move mouse to move the solid
now one direction works, but the other one not

maybe issue #158 is related with this one

mael25, 2021-07-30:
It is because when doing the Mouse+Shift click, it will change the `translation` field of the `Solid` and not the one of the `Transform` (regardless to the fact that they are defined within a Proto). Thus, although the translation of the `Solid` is modified as requested, the resulting movement appears strange because it is applied within the rotated frame of the `Transform`. You can observe it by right-clicking on the Proto --> Convert to Base Node(s).

Thomas-Feldmeier, 2021-07-30:
The Solid should move in the same direction of the mouse. As the coordinate system of the Solid can be rotated relative to the world coordinate system, the movement of the mouse has to be converted from the world coordinate system to the coordinate system of the Solid. In wb_supervisor_node_get_pose() such an transformation is done. Make a pose matrix with the mouse coordinate change between two updates as translation, rotation matrix should be identity matrix. Transform that matrix into the coordinate system of the Solid. Get the translation out of the new matrix, change solid position by this offset.

Thomas-Feldmeier, 2021-07-30:
maybe you need to invert the pose matrix [link](https://math.stackexchange.com/questions/152462/inverse-of-transformation-matrix)

Thomas-Feldmeier, 2021-07-30:
strg+mouse movement works correct (adding a force), that code could be used to fix shift+mouse movement.

closed 2021-08-04 by mael25