cyberbotics/webots #3529: `wb_supervisor_node_set_visibility` also hides sibling nodes

stefaniapedrazzi, 2021-08-02:
In case of such a structure:
```
Solid {
  children [
    Shape A {
      ..
    }
    Shape B {
      ..
    }
  ]
}
```

Hiding the node A from a camera will also hide node B.
But if I add a `Transform` parent to the `Shape` that should be hidden then works correctly.

I will inspect if this is a limitation of the functionality, but in this case it should be clearly specified in the documentation that currently simply says that children are recursively hiddden too:
https://www.cyberbotics.com/doc/reference/supervisor#wb_supervisor_node_set_visibility

stefaniapedrazzi, 2021-08-02:
Only geometry and transform nodes have their own WREN node.
All the other nodes (for example the Shape node) inherit the parent node, and thus if you set the visibility of a node that is not a geometry or a transform the visibility is also applied to all the siblings nodes.

omichel, 2021-08-02:
From a user's perspective it seems strange that it doesn't work for a shape, but works for a geometry. Usability should not be a consequence of the underlying implementation (here in WREN). Instead, I believe we could change this behavior so that when a Shape is provided, it works also (and internally, it will apply to its geometry).

stefaniapedrazzi, 2021-08-04:
Yes, I agree.
The current behavior seems buggy from a user's perspective and it would better to fix it.
However, I would not add an exception just for the `Shape` node but fix it properly for all the nodes (`Group`, `Slot`, joints, etc.) applying it recursively only to the children nodes.

PR #3543 (2021-08-09): Fix supervisor node set visibility
Fix #3529:
if given node doesn't have a dedicated Wren node, then apply visibility to closest descendants with dedicated Wren node.

Only `WbBillboard`, `WbGeometry`, `WbTransform` and `WbMuscle` have a dedicated Wren node.
For all the other nodes we have to retrieve the descendant nodes:

Nodes overriding the function to retrieve closes descendant nodes with dedicated Wren node:
|Node | Function Overridden | Comments |
| ------ | --------- | -------- |
| BasicJoint | x  | |
| Billboard | x | |
| Geometry | x | |
| Group | x | |
| Motor | x | |
| Muscle  | x  [... cut]
closed 2021-08-09 by stefaniapedrazzi